ms-iot/remote-wiring #25: Only works with 'OldStandardFirmata'

frogger3d, 2015-06-13:
It seems the library only works when you flash an Arduino with the 'OldStandardFirmata'
sketch. When you flash the StandardFirmata it connects fine, but nothing happens. It would be wise to add this to the documentation or better yet support the new standard sketch.

I haven't dug deeper, but this was using the https://github.com/ms-iot/windows-remote-arduino-samples project with an Arduino Uno connected to a Surface Pro 3 using a BlueSmirf Silver

frogger3d, 2015-06-13:
Oops, just a case of different defaults.. My bluetooth device was set to 115200 baud, which is the default for OldStandardFirmata. The StandardFirmata is defaulted to 57600 baud, but when changed to 115200 it worked just fine. The code comment in RemoteBlinky threw me off (baud rate does matter, but only on the Arduino side):

            //these parameters don't matter for bluetooth
            bluetooth.begin(115200, SerialConfig.SERIAL_8N1);

closed 2015-06-13 by frogger3d

turkycat, 2015-06-14:
I'll update that comment for clarity, thanks for following up!