ms-iot/remote-wiring #28: [RemoteWiring] Pinmode uses firmata constructs instead of wiring

connerbrooks, 2015-06-16:


zfields, 2015-06-24:
Should be `pinMode(A1, OUTPUT);`, not `pinMode(A1, PWM);`

zfields, 2015-08-13:
@turkycat Were you able to resolve this by providing multiple signatures for `pinMode()`?

turkycat, 2015-08-13:
I believe this issue represents the fact that pinMode accepts an enum type which includes ANALOG, PWM, SERVO, IGNORED, etc rather than just INPUT and OUTPUT. As it currently exists, these constructs are necessary to attach servos, differentiate between digital and analog input, etc

zfields, 2015-08-13:
I don't think the ability to use additional (more descriptive) pinmodes is the problem. I believe the issue was raised because Conner was not able to use `pinmode()` in the way it would be used with Arduino, therefore the library was unintuitive.

turkycat, 2015-08-13:
If the new overloads for pinMode accepting strings like "A0" solves this problem, then that's great, and we can close the issue. I assumed the "construct" in question was the extra (firmata-defined) "modes"

zfields, 2015-08-13:
You have a better grasp of the new library functionality than I do, and I was with Conner when he posted the issue. Let's whiteboard it to verify, but I think this solution may be sufficient.

------ I'll update the issue and respond below with the result of our meeting -------

turkycat, 2015-10-29:
@zafields ping

closed 2015-10-29 by zfields