ms-iot/remote-wiring #30: [Firmata] missing `void startSysex(void)` method

zfields, 2015-06-17:


ooeygui, 2015-08-11:
Can you supply information as to why it is needed?

turkycat, 2015-08-11:
Firmata.cpp calls it "startSysex" / "endSysex" while I named it "beginSysex" / "endSysex" in UwpFirmata. I agree that it should have matched the latter, but not sure if we should change it now?

zfields, 2015-08-13:
```
void FirmataClass::startSysex(void) 
{ 
    FirmataStream->write(START_SYSEX); 
} 
```
```
bool 
UwpFirmata::beginSysex( 
    uint8_t command_ 
    ) 
{ 
    _sys_command = command_; 
    _sys_position = 0; 
    return true; 
} 
```
These do not have similar functionality...

zfields, 2015-08-13:
We are trying to provide a familiar API surface, so we should at least have parity. I would take the logic from `beginSysex()`, move it `startSysex()`, then have `beginSysex()` call `startSysex()` and deprecate `beginSysex()`.

turkycat, 2015-10-29:
We should revisit this issue. New UwpFirmata implementation does not use `startSysex()` or `endSysex()`. New function signatures are:

```
[Windows::Foundation::Metadata::DefaultOverload]
void
sendSysex(
        SysexCommand command_,
        IBuffer ^buffer_
    );
```
and
```
void
    sendSysex(
        uint8_t command_,
        IBuffer ^buffer_
    );
```

which allow the entire IBuffer to be sent at once. No more need to invoke 3 functions for one concept. It does not perfectly match the Firmata API (as this issue was meant to resolve) but is arguably more sensible on our platform.

zfields, 2015-10-29:
I'll close this issue. The original observation is no longer valid.

closed 2015-10-29 by zfields