ms-iot/remote-wiring #37: Access Violation with USB connection on Raspberry Pi 2

Hijdra, 2015-07-01:
Hi,

I am trying to get a connection to my Arduino Micro with the UsbSerial solution but I keep getting 'Access Violation' errors after initialization. I use a Raspberry Pi 2 for running the code.

Below the code where it fails, I edited the sample code just a bit.

The code fails a half second after the Connection Established breakpoint is hit when the 'usb.begin(57600, SerialConfig.SERIAL_8N1)' code is called. It does not throw any exceptions and kills the app + debug session.

[image]

Side info:
My Raspberry Pi 2 is on OS version '10.0.10152.0'.
Works smoothly on as Windows 10 desktop app (build: 10158).

Greetings,
Mark Hijdra

Hijdra, 2015-07-03:
I have found the problem which is causing this.

After debugging an app the app is not successfully closing which means the app is still keeping the USB open.
When another app also wants to open the USB it gets the above crash.

Is there a way to close the USB connection?
When I try to execute 'usb.end()' the app is hanging on that line of code.

Greetings,
Mark Hijdra

ooeygui, 2015-08-11:
Thank you for the report. We will look into it.

amitchaudhary, 2015-08-29:
Run VS with Administrator mode to fix the issue.

turkycat, 2015-10-29:
Hey @Hijdra have you tried amit's suggestion? You may also be interested in using the latest codebase with our current RTM build 10240 (or current insider build 10556). There have been several improvements and bug fixes since August!

Hijdra, 2015-11-06:
Hi @turkycat, sorry I haven't worked with IoT since then, maybe I'll give the new version a try soon.

turkycat, 2015-11-24:
Hello @Hijdra! I'm going to close this issue as the code you were using originally is quite out of date and USB has been improved since then! Feel free to try again and create a new issue if you are still experiencing difficulties :smile:

closed 2015-11-24 by turkycat