ms-iot/remote-wiring #46: analogRead(string) fails

ManniAT, 2015-08-15:
When I call remDevice.analogRead("A0"); I get an error result.

Debugging the source I found that in Line 132 _analog_offset is added.
For my case this means "A0" - becomes 18
analogRead(18) is called.
And there at line 99 the _analog_offset is added again...

90 uint16_t 
91 RemoteDevice::analogRead( 
92     uint8_t pin_ 
93     ) 
94 { 
95     //critical section equivalent to function scope 
96     std::lock_guard<std::recursive_mutex> lock( _device_mutex ); 
97 
 
98     uint16_t val = -1; 
99     uint8_t analog_pin_num = pin_ + _analog_offset; 
100 
 
101     if( _pin_mode[ analog_pin_num ] != static_cast<uint8_t>( PinMode::ANALOG ) ) 
102     { 
103         if( _pin_mode[ analog_pin_num ] == static_cast<uint8_t>( PinMode::INPUT ) ) 
104         { 
105             pinMode( analog_pin_num, PinMode::ANALOG ); 
106         } 
107         else 
108         { 
109             return static_cast<uint16_t>( val ); 
110         } 
111     } 
112 
 
113     if( pin_ < _num_analog_pins ) 
114     { 
115         val = _analog_pins[ pin_ ]; 
116     } 
117 
 
118     return val; 
119 } 
120 
 
121 uint16_t 
122 RemoteDevice::analogRead( 
123     Platform::String^ analog_pin_ 
124     ) 
125 { 
126     uint8_t parsed_pin = parsePinFromAnalogString( analog_pin_ ); 
127     if( parsed_pin == static_cast<uint8_t>( -1 ) ) 
128     { 
129         return static_cast<uint16_t>( -1 ); 
130     } 
131 
 
132     return analogRead( static_cast<uint8_t>( parsed_pin ) + _analog_offset ); 
133 } 
134

turkycat, 2015-08-16:
> 101 if( _pin_mode[ analog_pin_num ] != static_cast( PinMode::ANALOG ) )
> 103 if( _pin_mode[ analog_pin_num ] == static_cast( PinMode::INPUT ) )
> 109 return static_cast( val );
> 127 if( parsed_pin == static_cast( -1 ) )
> 129 return static_cast( -1 );
> 132 return analogRead( static_cast( parsed_pin ) + _analog_offset );

The analog pins on Arduino and similar devices are numbered after the
digital pins, so "A0" should become 14 on an Arduino uno because there are
14 digital pins numbered 0-13, hence the analog offset. What type of device
are you using?

Jesse

ManniAT, 2015-08-16:
Hi Jesse,

I know what you mean - an my Yun A0 is pin 18.
Try to examine the code...

I call analogRead("A0").
In this method parsed_pin becomes correctly 0 - then (on line 132) analogRead is called with parsed_pin + _analog_offset (which is correctly 18 on my yun).
So this results in the call analogRead(18)

At line 99 the _analog_offset is again added.
So _analog_pin_num becomes 36 - FAIL.

ManniAT, 2015-08-16:
Just to be more precise - it doesn't matter if the offset is 14 or 18 or whatever.
The offset is added twice this is the problem

zfields, 2015-08-17:
HI ManniAT,

As a work around, you can use pin 18 directly on your call to `pinMode(18, PinMode::ANALOG)`, then read it using `analogRead(0)`.
It is confusing, but it reflects the underlying implementation of Firmata, and it will work for you until we get a hot fix out the door. We are glad you're using the library, and thanks for reporting the issue!

ManniAT, 2015-08-17:
Hi zfields,

I simply changed line 132 from
return analogRead( static_cast( parsed_pin ) + _analog_offset );
To
return analogRead( static_cast( parsed_pin ) );

And everything works as expected.

zfields, 2015-08-17:
Awesome news! Have you submitted a pull-request?

ManniAT, 2015-08-17:
Ok, I did it.. Request is submitted

closed 2015-08-18 by ManniAT