ms-iot/remote-wiring #49: Ethernet Example

davidwallis3101, 2015-08-25:
I'm trying to get the ethernet connection working..

in stdfirmata ethernet I have local IP set as a valid ip for the shield, remote IP I have as the ip for the raspberry pi? (correct way around?) - If I set the remote IP to my PC I can see it attempting to connect on port 3030  (I would have thought the arduino would be the socket server?)

anyway this is the code I am using, is it correct?

        private async Task MicrocontrollerTest()
        {
            Debug.WriteLine("Arduino task running");
            try
            {
                // 192.168.0.55 is the IP assigned in firmata as LocalIP - this correct?
                IStream connection = new NetworkSerial(new HostName("192.168.0.55"), 3030);

                RemoteDevice arduino = new RemoteDevice( connection );

                // Add Handlers for connecton states
                connection.ConnectionEstablished += OnConnectionEstablished;
                connection.ConnectionFailed += OnConnectionFailed;

                // Add handler to configure pins
                arduino.DeviceReady += Setup;

                //subscribe to the DigitalPinUpdateEvent with the name of the function to be called.
                arduino.DigitalPinUpdated += MyDigitalPinUpdateCallback;

                // Params ignored in underlying C++ code.
                connection.begin(115200, SerialConfig.SERIAL_8N1);

            }
            catch (Exception ex)
            {
                Debug.WriteLine("Error " + ex.Message);
                throw;
            }
        }

davidwallis3101, 2015-08-25:
this is the firmata config: 

#define remote_ip IPAddress(192, 168, 0, 48)  // PI

// *** REMOTE HOST IS NOT YET WORKING ***
// replace with hostname of server you want to connect to, comment out if using 'remote_ip'
// #define remote_host "server.local"

// STEP 3 [REQUIRED unless using Arduin Yun]
// Replace with the port that your server is listening on
#define remote_port 3030

// STEP 4 [REQUIRED unless using Arduino Yun OR if not using DHCP]
// Replace with your board or ethernet shield's IP address
// Comment out if you want to use DHCP
#define local_ip IPAddress(192, 168, 0, 55)

// STEP 5 [REQUIRED unless using Arduino Yun]
// replace with ethernet shield mac. Must be unique for your network
const byte mac[] = {0x90, 0xA2, 0xDA, 0x00, 0x53, 0xE5};

turkycat, 2015-08-26:
Hey David! Thanks for submitting the issue and letting us know of this incompatibility!

As it turns out, the StandardFirmataEthernet sketch is a recent addition to the Firmata library, and was designed with the intention of running as a client and connecting to a server. As you may have noticed, the sketch itself calls out the fact that it doesn't yet support running as a server and accepting clients.

The current design of networking in WRA intended for the opposite, the Arduino runs the server and the library acts as a client to the device. We are currently looking into the investment to support StandardFirmataEthernet, but for now you must use a different sketch to enable wifi or ethernet support.

Please refer to the section of the readme which goes into further detail: https://github.com/ms-iot/remote-wiring#notes-on-wifi-and-ethernet

Here is the repository which contains the required libraries and sketches for both standard-firmata-ethernet and standard-firmata-networking which will work with the existing WRA library (tested again just now to verify!): https://github.com/turkycat/standard-firmata-networking

davidwallis3101, 2015-08-26:
Thanks, I will try that from the github repository listed, I will then post back with an example solution :)

turkycat, 2015-10-29:
Hey @davidwallis3101 have you had any success with the above? Let me know if I can close your issue :)

davidwallis3101, 2015-11-01:
Yeah close the issue, all good

closed 2015-11-04 by ooeygui