ms-iot/remote-wiring #51: DigitalPinUpdated Callback not working?

davidwallis3101, 2015-08-28:
I've now got the device communicating over ethernet, but I seem unable to get the DigitalPin Callback events to work, analog works ok, but is getting fired a lot

If I use remote commander from my tablet I can see the values change on the digital input pin (7)

Is this a known issue (i'm using arduino uno boards (possibly a clone - but works in remote commander so I dont think its that.)

ooeygui, 2015-08-28:
Thank you for reporting. @turkycat will take a look.

davidwallis3101, 2015-08-28:
Hmm Works on Pin 8, not on pin 7..

turkycat, 2015-08-28:
Hey David!
I've tried this just now and it is working fine for me on all pins (except for shield pins as described below). Are you using an official Arduino Ethernet shield or some other derivative? Some pins are in use when using certain shields, but the Ethernet shield should use only 10, 11, 12, 13 and when using SD card, 4.

davidwallis3101, 2015-08-29:
Given it was cheap of ebay I suspect its a derivative, it says HanRun HR911105A on the RJ45, it has the Wiznet5100 controller and an SD card on it.

It looks like they have marked 10 - 13 as being used.

David

davidwallis3101, 2015-08-29:
interestingly I try it again today after configuring all the pins and tested it and they all seem to now work apart from pin 2.. hmm.

Oh well at least I'm making progress and its not an issue.. as for getting usb working, Ill save that for another day :)

closed 2015-08-29 by davidwallis3101