ms-iot/remote-wiring #54: Arduino AnalogRead with a LM35 temp sensor - Always returns 65535

VannosG, 2015-09-13:
I've got an Arduino successfully connected and the RemoteBlinky app working. I've setup the Arduino A0 pin with `arduino.pinMode("A0", PinMode.ANALOG);` and i'm reading it with `UInt16 rawTemp = arduino.analogRead("A0");` but `rawTemp` is always 65530.

The LM35 is connected to 5V, GND and A0 on the Arduino (this setup works correctly using the sample Arduino sketch.) Using my DMM I can see the chip is correctly putting out 0.2V to the A0 pin.

Anyone have any idea whats going on? Thanks!

VannosG, 2015-09-13:
I've resolved my issue, but i'm not sure what the root cause was, something threading related I think.

In my `OnConnectionEstablished()` method (copied from the RemoteBlinky demo) I was setting my pin modes. I moved those pinmode setup calls out into the handler of a periodic thread pool timer which I had running to read the temp.

turkycat, 2015-09-18:
Were you originally attempting to read the value of the sensor immediately after setting the mode to ANALOG? Due to the way Firmata works, it takes a short time (max 19ms + [ 2x latency ] ) before the value is reported and available for reading. This might explain why the other thread is able to read it, as there is likely some delay there before the attempt is made. You can always subscribe to the AnalogPinUpdated event to be notified immediately (and every time) a value is reported by the device.

turkycat, 2015-11-24:
Closing this issue. A bug was found at one point that actually did cause this issue in certain circumstances, and has been fixed. Thank you for helping to alert us to the problem! If you experience this problem or any other using the latest code, please open a new issue and let us know!

closed 2015-11-24 by turkycat