ms-iot/remote-wiring #57: "sendSysex" doesn't work properly

EdgarSun, 2015-09-21:
I followed this tutorial https://github.com/ms-iot/remote-wiring/blob/develop/advanced.md to setup a very simple custom command. "FirmataConnectionReady" works properly when initializing Firmata the first time, but Firmata.sendSysex doesn't work when sending custom command. below are the codes:

On Arduino Side - >
#define ALL_PINS_COMMAND 0x42
void sysexCallback(byte command, byte argc, byte *argv)
{
  byte mode;
  byte slaveAddress;
  byte data;
  int slaveRegister;
  unsigned int delayTime;

  switch (command) {
    case ALL_PINS_COMMAND:
      Firmata.sendString("Hello from arduino!");
    break;
  ... ...
  ...

C# Side - >

const byte ALL_PINS_COMMAND = 0x42;
... ...
public void SendSysexCommandHandler(object arg)
{
    if(Firmata.connectionReady()) <----------- return true
    {
      byte b = 1;
      Firmata.sendSysex(ALL_PINS_COMMAND, new byte[] { b }.AsBuffer());
    }
}

turkycat, 2015-09-21:
Hey EdgarSun! can you clarify what behavior you are seeing when you run the above? Are you sending the sysex but not receiving the string reply?

For debugging, I would suggest:
1) Verify that you have subscribed to the StringMessageReceived event in RemoteDevice. This event will be invoked when a `sendString` is received from the device.
2) Modify the Arduino sketch to use `digitalWrite` to manually light an LED when ALL_PINS_COMMAND is received. This will help us identify if the trouble is with `sendSysex` or with the string reply.

EdgarSun, 2015-09-22:
I put some debug codes in sysexCallback, it works well when initializing firmata the first time, because I can got the command message from "StringMessageReceived" event handler in c#.

Logs:
Received string message from arduino: 240
Received string message from arduino: 107
Received string message from arduino: 107
Connected arduino!

[image]

[image]

The problem is there was not any command message received when when running below codes:
[image]

For debugging:
1) Verify that you have subscribed to the StringMessageReceived event in RemoteDevice.
    -> Yes, this event's hooked up correctly, I can get string message at the beginning when initializing firmata

2) Modify the Arduino sketch to use digitalWrite to manually light an LED
    -> Sorry, I don't have a LED for testing digitalwrite. But it seems the function "sysexCallback" was not triggered any longer when executing "Firmata.sendSysex()" on c# side.

turkycat, 2015-11-24:
So you are getting some messages from the Arduino when it first initializes, but you are never seeing the "Hello from Arduino" string come back when a sysex message is sent to the Arduino ( and it is assume this code should be invoked on Arduino side)?

```
switch (command) {
case ALL_PINS_COMMAND:
Firmata.sendString("Hello from arduino!");
break;
```

We want to identify the point of failure... and I think its a bit weird that your string data is 240, 107, 107 since 240 doesn't correspond to an ascii character...so lets first make sure you can receive strings at all. Add this around your `loop()` section:

```
long lastStringMessageSent = millis();

loop()
{
  if( ( millis() - lastStringMessageSent ) > 1000 )
  {
    Firmata.sendString( "Ping from Arduino!" );
    lastStringMessageSent = millis();
  }
  
  //... (rest of loop)
}
```

This will send a "heartbeat" ping every 1 second with a string message. Once you get your app connected to the Arduino, you should see "Received string message from arduino: Ping from Arduino!" printed every 1 second.

If that works, add a `Firmata.sendString` call that is invoked when the `sysexCallback` is first called. If that works, but you still don't see the "Hello from Arduino" message coming through, then something may be wrong with the way ALL_PINS_COMMAND is defined.

TAMHAN, 2016-03-20:
Hello,
I wanted to confirm that I have similar issue - Firmata sending works for about a second, and then stops.

Amusingly, receiving data sent from Arduino to PC works ad perpetuum.

frogger3d, 2016-03-21:
Pull request #92 seems to fix this. To work with the current version (1.3) you should add a frush() call after each sendSysex.

IoTGirl, 2017-03-20:
Closing based on above pull request

closed 2017-03-20 by IoTGirl