ms-iot/remote-wiring #58: Multiple Bluetooth Connection Supported?

amitchaudhary, 2015-09-22:
The scenario is…
- There is one Windows 10 IoT Raspberry Pi gateway
- Two Arduino UNO devices connected with gateway, where Gateway controls them via using “Windows Remote Arduino” especially reading the sensor value connected with both Arduino.
-        The communication between Gateway and other devices (slaves) is via Bluetooth. 
- Gateway is controlling the both Arduino via Windows Remote Arduino using “Firmata”.

Now, problem is, even though I'm making two different Bluetooth connection with Arduino from Raspberry Pi, still only one able to read the sensor/Arduino value?
So, seems like there is issue when making multiple connection? Isn't is supported scenarios/what? Please help.

amitchaudhary, 2015-09-28:
No response?

turkycat, 2015-10-28:
I'm very sorry for the late response. I remember discussing this issue with our team and must have forgotten to respond to you. 

The design of WRA absolutely allows for multiple connections. There is nothing bound between different IStream objects (each is completely unique). It was very intentional from the start to allow multiple devices to be connected through multiple different methods.

However, there seems to be some difficulty when connecting to multiple Bluetooth devices using RFCOMM technology. I have just tested this again using two different Bluetooth SPP devices connected to two different Arduinos. The library itself handles this just fine, but I am actually seeing an internal failure inside the RFCOMM dlls. I'm going to reach out to some contacts within MS and investigate this further. It may be that the scenario of two SPP devices is fundamentally not supported within Windows, but it should be possible according to this page: https://en.wikipedia.org/wiki/List_of_Bluetooth_protocols#Radio_frequency_communication_.28RFCOMM.29

erviveksoni, 2016-03-21:
Any update on this one?

jsiegmund, 2016-04-20:
I'd like that update too :)

jase147, 2017-02-15:
Hello. Will this be supported? Thanks.

saraclay, 2018-08-21:
If this issue is still relevant, please file this at: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-08-21 by saraclay