ms-iot/remote-wiring #63: UsbSerial.Read outputs randomly crap

Jiihaa, 2015-10-08:
I'm wondering if there's something wrong with my code or the UsbSerial.Read -function, as I have problem that it outputs crap randomly messing my output. I have tried different baud rates as well, with the same problem.

I'm creating the connection like this:
connection = new UsbSerial(selected);
connection.ConnectionEstablished += Connection_ConnectionEstablished; ;
connection.begin(9600, SerialConfig.SERIAL_8N1);

and in the connectionEstablished I'm reading the serial like this:

while (true)
{
    Debug.Write(Convert.ToChar(connection.read()));
}

My Arduino code writes in a loop 1234567890 and that's how the output shows in Putty for example, but in Visual Studio debug output I can see that it does this every 5-8 line:1�234567980

Any pointers if I'm doing something wrongly?

turkycat, 2015-10-08:
The default baud rate for firmata is 57600, not 9600. You should change it in one of the two locations so that they match

Jiihaa, 2015-10-09:
I tried it with 57600, it has no effect on the weird behaviour

turkycat, 2015-10-28:
Looking into this issue further. You are reading directly from the connection object, which will return -1 if there is no data available. It is very likely that your Windows device will run much faster than the device sending the data. Can you modify your while loop to the following and advise if the issue still exists?

```
while (true)
{
  ushort val = connection.read();
  if( val != ((ushort)-1) )
  {
    Debug.Write(Convert.ToChar(val));
  }
}
```

turkycat, 2015-10-28:
you may also use this, which is more typical:
```
while (true)
{
  if( connection.available() > 0 )
  {
    Debug.Write(Convert.ToChar(connection.read()));
  }
}
```

turkycat, 2015-11-24:
Closing this issue as I have been unable to replicate the issue. If you are still having issues with the latest codebase, you may open a new issue!

closed 2015-11-24 by turkycat