ms-iot/remote-wiring #67: arduino.analogRead() and arduino.digitalRead() cannot work on Leonardo with Firmata

jimaobian, 2015-10-21:
1. arduino.analogRead() and arduino.digitalRead() **cannot** work on **Leonardo** with Firmata
2. arduino.analogRead() and arduino.digitalRead() **can** work on **UNO** with Firmata
3. The Serial TX LED doesn't blink on Leonardo, which indicates that no data sends from Leonardo to Raspberry Pi.
4. If running Processing on PC, analogRead() and digitalRead() **can** work on **Leonardo** with Firmata.

PC: 
Windows 10 RTM
VS Community 2015

Raspberry Pi 2:
Windows 10 IOT RTM

Device:
Arduino Leonardo

remote-wiring version:
using the latest :
Install-Package Windows-Remote-Arduino

jimaobian, 2015-10-26:
https://github.com/ms-iot/remote-wiring/pull/69
This Patch fix it

jimaobian, 2015-10-26:
This patch may also fix the analogRead and digitalRead problem on DFRobot Bluno and Bluno Bettle.

turkycat, 2015-10-28:
Your patch does resolve this issue. Please comment if we may close this!

jimaobian, 2015-11-04:
Close

closed 2015-11-04 by jimaobian