ms-iot/remote-wiring #70: Arduino DUE & Bluetooth on Serial2

akprog, 2015-10-26:
Is it possible to use Bluetooth module connected to Serial2 on Arduino DUE with this firmata version?

turkycat, 2015-10-28:
Yes, although we haven't explicitly tested that configuration. You will need to tell Firmata to use Serial2 instead of the default Serial. Update your StandardFirmata sketch as follows:

locate this line in setup():
```
  Firmata.begin(57600);
```

and change it to the following:
```
  Serial2.begin( 57600 );  //use your Bluetooth device's baud rate here. Must match the device
  Firmata.begin( Serial2 );
```

akprog, 2015-10-30:
@turkycat, thanks for your reply! I changed the Arduino sketch as suggested - but with no effect.
I use a HC-06 Bluetooth module and it stops blinking when the App is connected to the Arduino. But when I try to do arduino.digitalWrite(53, PinState.LOW); (Pin 53 is where an LED is connected).

turkycat, 2015-10-30:
What happens when you try the above? I think you left your reply unfinished. It does sound like your device is getting a connection, but these types of problems are hard to debug. Try hooking up another led to a different pin and modify your sketch to turn that led on when the digital write callback is invoked. If that doesn't light, try turning it on at earlier and earlier points until you can find the point of failure

akprog, 2015-10-31:
@turkycat Thanks again for your advices. My current setup looks like this:

**The Arduino DUE setup:**
Opened StandardFirmata example and changed setup() like tihs

    Serial2.begin( 57600 );  //use your Bluetooth device's baud rate here. Must match the device
    Firmata.begin( Serial2 );
    // Setup DigitalPin 53
    pinMode(53, OUTPUT);
    digitalWrite(53, HIGH); // Start with LED on

That's all on Arduino side.

In **Visual Studio** I followed the instructions here: https://ms-iot.github.io/content/en-US/win10/SetupPCWRA.htm
Then connect to Arduino like this:

    connection = new BluetoothSerial("HC-06");
    connection.begin(57600, SerialConfig.SERIAL_8N1);
    arduino = new RemoteDevice(connection);

And implement a Button to set Pin 53 to LOW:

    private void button1_Click(object sender, RoutedEventArgs e)
        {
            arduino.digitalWrite(53, PinState.LOW);
        }

A I mentioned earlier, the HC-06 module stops blinking but I can not set Pin 53 to low.

I also used a Bluetooth-Terminal for windows to send Strings via Bluetooth-Com-Port to arduino. I can see the incoming strings on Arduino when I open the Serial Monitor, so bluetooth seems to work fine. I did some other small projects and sent commands via Com-Port, but in this case I would like to use the Firmata benefits :)
Thats my status quo.

turkycat, 2015-11-05:
I've ordered an HC-06 device so that I can test your setup. In the meantime, I noticed that the device that I ordered has a default baud rate of 9600. Yours may be different, but if the HC-06 is not configured to use 57600, it will not work with the setup you currently have. Have you checked with the Bluetooth device's documentation to verify that either 1) the default baud rate is 57600 or 2) you have gone through the necessary steps to change the baud rate to 57600?

akprog, 2015-11-06:
Some days ago I have tested a second setup (HC-06 on Arduino UNO's Serial port) with baud rate 9600 and it works fine. I have changed the DUE (Serial2) setup to 9600 but it still dont work (no blinking RX when I try to send data). Do you have a running setup with HC-06 on SerialX?
Please also let me know if you need additional hardware, maybe I can ship it to you so you don't have to order (and pay) it.

turkycat, 2015-11-06:
No current setup with HC-06, but I will try when it is delivered to me. I have successfully used SerialX instead of `Serial` in other sketches with no problems, I've even written my own `Stream` classes (like `Serial`) and used them in place of `Serial`

IoTGirl, 2017-06-20:
Closing Stale - No post from originator or owner in more than a year.

closed 2017-06-20 by IoTGirl