ms-iot/remote-wiring #72: Channel security

MrSmersh, 2015-11-02:
As far as I see the communication channel between Arduino and phone is not secure. Could you do something about it? In some scenarios this could be a real pain issue since someone could hack the the communication and control the Arduino from a "spurious" source. Like the first generations TV remotes, the neighbor could control your Arduino.

turkycat, 2015-11-24:
We have no plans at this time to add encryption or device security. It is expected that devices would need to be either paired (via Bluetooth), physically connected (via USB or Ethernet cable), or authenticated to a wireless network in order for connections to be made to them. If you still desire additional security, I would advise building on top of or modifying RemoteDevice (or perhaps build a new secure RemoteDevice implementation on top of the Serial and Firmata layers).

turkycat, 2015-11-24:
An amendment to the above: we will take this on as an action item in our backlog and investigate the security improvements in the future. It is still true that you could amend the existing code if you need improved security right away!

saraclay, 2018-08-21:
If this issue is still relevant, please file this at: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-08-21 by saraclay