ms-iot/remote-wiring #78: unable to write or read data from arduino

abhishekdaya, 2015-11-08:
I am using arduino uno and i am able to connect to arduino via bluetooth but I am un able to read or write data to the device... Pls help.. I have tried changing the TX RX of bluetooth with aurdino both ways... My bound rate is 57600 in firmata and bluethooth....

abhishekdaya, 2015-11-08:
I am sorry to Arduino not from Arduino (in the subject)

akprog, 2015-11-09:
Hi abhishekdaya, can you please share more information about your setup?

turkycat, 2015-11-09:
Hey @akprog , I noticed you haven't mentioned which Bluetooth device you are using. Have you checked that the device both supports and is *currently configured* to use 57600? This is a special consideration you must make when using Bluetooth.

abhishekdaya, 2015-11-12:
I figured this out... Thanks...

closed 2015-11-12 by abhishekdaya