ms-iot/remote-wiring #79: example of digitl pin updated

abhishekdaya, 2015-11-12:
Hi... Can anyone please show an example of digitalpinupdated and analogpinupdated code.... Tnks....

abhishekdaya, 2015-11-12:
I just have the folowing queries...
1) Digital Update only gives the state when changed from Low to High...
2) The above is happening only once.... I.E. The second time i toggle a switch i dont get any feedback.....

abhishekdaya, 2015-11-12:
I am using arduino uno and the commincation is via bluetooth... I am using the module HC-05... @11520 BR...

turkycat, 2015-11-24:
Hey @abhishekdaya , we've got another report of an issue with HC-06 devices. It may be a problem with similar devices as well. I've got an HC-06 on order and will update the issues once I have been able to test the device

zfields, 2015-12-05:
@abhishekdaya Can you provide a pointer to your code?

IoTGirl, 2017-06-20:
Closing stale - No response in more than a year.

closed 2017-06-20 by IoTGirl