ms-iot/remote-wiring #80: Pin configuration not received

EdgarSun, 2015-11-16:
Hi,

I am testing "sendSysex" function between windows phone 8.1 and arduino (with SparkFun Bluetooth Mate Silver). the connection can be established successfully, but I got below error message when executing "sendSysex" to turn on/off LED (https://github.com/ms-iot/remote-wiring/blob/develop/advanced.md).  
```
Arduino Failed: A device connection was established, but the device failed handshaking procedures. Verify that your device is configured with StandardFirmata. Message: Catastrophic failure
Pin configuration not received.
```
(StandardFirmata version: 2.5)

ooeygui, 2015-11-16:
Thank you for reporting. We will set up the repro and see what's going on. (looks like the handshake process has changed)

EdgarSun, 2015-11-18:
Below are some codes for initializing the connections:
```
public const int CONN_BAUD = 115200;
... ...
...
DeviceStream.begin(CONN_BAUD, SerialConfig.SERIAL_8N1);
Firmata = new UwpFirmata();
Arduino = new RemoteDevice(Firmata);
Firmata.begin(DeviceStream);
Firmata.startListening();
```
Below are the sample codes for processing sysex command in arduino:
```
void sysexCallback(byte command, byte argc, byte *argv)
{
  byte mode;
  byte slaveAddress;
  byte data;
  int slaveRegister;
  unsigned int delayTime;

  switch (command) {
    case ALL_PINS_COMMAND:
    if( argc == 1 ) //verify we received the whole message
    {
        //the MSB is always sent in the LSB position 2nd byte
        byte val;
        if( argv[0] == 1 )
        {
            val = HIGH;
        }
        else
        {
            val = LOW;
        }

        //set the pins! On many Arduino devices (Uno, Leo, Yun, etc) there are 14 digital pins from 0 to 13.
        for( byte i = 0; i < 14; ++i )
        {
            digitalWrite( i, val );
        }
    }
    break;
... ...

```

turkycat, 2015-11-19:
The error you are getting actually isn't from the sysex functionality you're trying, its from the handshaking procedure that occurs when the initial connection is established. I can confirm that in `StandardFirmata` version 2.5 the author added a new pin mode which will break the current handshaking procedure. I am implementing a more foolproof fix for this now, but you can make a one line change to the file `RemoteDevice.cpp` (or revert to using `StandardFirmata` version 2.4.4 or 2.4.3).

In file `RemoteDevice.cpp`
line 522:
```C++
                i += 4;
```
change to:
```C++
                i += 6;
```

turkycat, 2015-11-19:
I've added the support for this new pin mode (and one other), but as I did so I realized that my original implementation allowed for additional data to be present without causing an error. This means that your handshaking failures are not due to the version change. Therefore, I wonder if you are using USB, Bluetooth, or networking to establish your connection? Have you validated your baud rate in StandardFirmata matches the value you are using above (`115200`)?

EdgarSun, 2015-11-19:
### I wonder if you are using USB, Bluetooth, or networking to establish your connection?
I am using bluetooth to establish connection between windows phone 8.1 and arduino.
> Bluetooth Device: SparkFun Bluetooth Mate Silver.

And I test the connections on three different arduino board, all of them have the same error messages: 
```
Arduino Failed: A device connection was established, but the device failed handshaking procedures. Verify that your device is configured with StandardFirmata. Message: Catastrophic failure
Pin configuration not received.
```
1) Arduino UNO
2) DFRobot Bluno v2.0
3) DFRobot Romeo v2.2 R3

And I tried to revert to firmata 2.4.4 and involved the changes (i += 6;) in RemoteDevice.cpp, but the error messages still exist. 

### Have you validated your baud rate in StandardFirmata matches the value you are using above (115200)?
Yes, I am using the same baud rate (115200) on both side

Below are the complete codes for setting up connections:

Arduino
```
Firmata.begin(115200);
while (!Serial) {
; // wait for serial port to connect. Only needed for ATmega32u4-based boards (Leonardo, etc).
}
```

C#
```
        var deviceStream = new BluetoothSerial(device); //device -> DeviceInformation
        deviceStream.begin(115200, SerialConfig.SERIAL_8N1);
		var firmata = new UwpFirmata();
		var arduino = new RemoteDevice(firmata);
		firmata.begin(DeviceStream);
		firmata.startListening();
```

turkycat, 2015-11-19:
I've tested using your code & the same configuration and having no trouble. I also tested using Firmata 2.5 and the current (unchanged) version of WRA. The error means that despite getting a connecting to the Bluetooth Mate Silver, it doesn't seem like any of the data is being sent/received properly between the two devices, or what is being sent/received is not understood.

How are you connecting the Bluetooth Mate Silver to the Arduino?

Will you also try this? I've rearranged your function calls a bit. The library is written so that the way you are doing it shouldn't cause a problem (and it didn't for me), but if it does for you I'd like to know about it. I've moved the IStream `begin` to be the last thing invoked, and I've removed `startListening`. `startListening` is automatically invoked by the RemoteDevice class when the device connection is established.

```C++
        var deviceStream = new BluetoothSerial(device); //device -> DeviceInformation
        var firmata = new UwpFirmata();
        var arduino = new RemoteDevice(firmata);
        firmata.begin(DeviceStream);
        deviceStream.begin(115200, SerialConfig.SERIAL_8N1);
```

EdgarSun, 2015-11-20:
The events of deviceStream (IStream) were hooked up with handlers in my original codes, and the handshaking issues was resolved and the "DeviceReady" event of arduino also was triggered successfully after removing below three event handlers.
```
deviceStream.ConnectionEstablished += DeviceStream_ConnectionEstablished;
deviceStream.ConnectionFailed += DeviceStream_ConnectionFailed;
deviceStream.ConnectionLost += DeviceStream_ConnectionLost;
```

Currently, I have another issue the LED will not be turned on/off when executing "sendSysex" function with below codes, it looks like the sysexCallback did not recevice command in arduino. Is there any way I can debug if the connection of firmata is available? 

```
[CommandHandle]
public void ToggleLED()
{
    byte ALL_PINS_COMMAND = 0x42;
    //we're defining our own command, so we're free to encode it how we want.
    //let's send a '1' if we want the pins HIGH, and a 0 for LOW
    byte b;
    if (_LEDState==false)
    {
        b = 1;
    }
    else
    {
        b = 0;
    }

    _LEDState = !_LEDState;
    //invoke the sendSysex command with ALL_PINS_COMMAND and our data payload as an IBuffer
    BluetoothHelper.Instance.FirstDevice.Firmata.sendSysex(ALL_PINS_COMMAND, new byte[] { b }.AsBuffer());
}
``` 

Below is a picture for demonstrating the device connections:

[image]

EdgarSun, 2015-11-21:
I made another test to see whether the sysexCallback was called properly with below temporary codes:
```
bool ledState=false;
void sysexCallback(byte command, byte argc, byte *argv)
{
  byte mode;
  byte slaveAddress;
  byte data;
  int slaveRegister;
  unsigned int delayTime;

  //-----------------------------------------------------
  if(ledState){
    digitalWrite( 13, LOW);
  }
  else{
    digitalWrite( 13, HIGH);
  }
  ledState = !ledState;
  //-----------------------------------------------------

  switch (command) {
    case ALL_PINS_COMMAND:**
  ... ...
```

The testing result is LED was lighted on when establishing the connection, it means "sysexCallback" was called at least one time. But when executing "sendSysex" to send command to arduino, there was nothing happened with the LED. I assume "sysexCallback" had nother response to the Sysex command.

turkycat, 2015-11-24:
Hey @EdgarSun , I replied to your other issue thread on `sendSysex` issues. I'll leave this issue open for now as our conversation continues, but I believe we are now discussing the same issue on both threads, correct?

EdgarSun, 2015-11-25:
@turkycat, Yes, you are right. the original issue of this thread has been resolved. my last reply is a same issue I reported in another thread [Duplicate Issue](https://github.com/ms-iot/remote-wiring/issues/57). the only tiny difference is I was testing "sendSysex" with USB connection between raspberry pi 2 and arduino a few weeks ago. and I meet a same issue when testing with bluetooth connection between windows phone 8.1 and arduino.

pbalut, 2016-04-05:
I have Arduino Mega 2560 and still the same error: 

> Arduino Failed: A device connection was established, but the device failed handshaking procedures. Verify that your device is configured with StandardFirmata. Message: Catastrophic failure
> Pin configuration not received.

**Configuration:**
Windows 10, connection via USB, baud rate set on 115200 (device drivers)
Firmata 2.4.4
Repo cloned from [this ](https://github.com/ms-iot/windows-remote-arduino-samples)tutorial
I tried all below configurations: 
```
   Serial1.begin(115200);
   Firmata.begin(Serial1);
```
```
  Firmata.begin(115200);
  while (!Serial) {
  / ; // wait for serial port to connect. Only needed for ATmega32u4-based boards (Leonardo, etc).
  }
```
```
Serial1.begin(115200);   //or your baud rate here, it will be 115200 if using the Bluetooth Mate Silver or Gold
while( !Serial1 );
Firmata.begin( Serial1 );
```

Could you help me whit this issue ? Thanks !

razvan252, 2016-04-18:
Hello. Same exact issue as  pbalut above. Mega 2560 with standard firmata sketch. Pin configuration not received and it says it's failing the handshaking procedures.

Can somebody please help?

pbalut, 2016-04-18:
I bought Arduino Uno and now everything works great...

razvan252, 2016-04-19:
From what I can gather the Mega isn not replying back after the initial request. Either the sketch needs updating or the request is wrong.

I do not wish to switch to Uno :\

razvan252, 2016-05-02:
Coming back to this issue, I had a bit of time to debug it on the mega and partially solved the problem. From what I can tell, the mega has a lot more pins than the Uno and as such, takes a bit longer to count them all and build the capabilities. Thus, it would send a new request before giving the board time to complete the request which was sent before and never giving it time to actually respond.

In RemoteDevice.cpp, around line 497, I changed Sleep( delay_ms ); to Sleep( 1000 );
Obviosuly it is a temporary solution but at least it works.

mcinnes01, 2016-05-15:
Hi,

I am getting this same issue intermittently with the Arduino Nano cat feeder I've built. To test the connection I've tried using the remote Arduino experience app from either my laptop or phone (both win 10). The only way I've found to resolve the issue is to restart my device. Sometime this works straight away other times a couple of restarts fixes it. I've had it running reliably for about a week, but then it bombed out again. I'm running my own dhcp server from my physical server, but as I haven't setup a secondary dhcp server sometimes I get issues. I wonder if this issue could be related to a network problem? 

I haven't tried the timeout idea but I can give that a go to try to rule that out.

Out of interest does the standard firmata ethernet image retry to connect if it fails for something like a dhcp issue?

Many thanks

Andy

mcinnes01, 2016-06-27:
Hi,

To add more to this, I am seeing this issue repeatedly now in a number of situations.

I've seen the problem intermittently with my cat feeder using a nano and W5100 ethernet module with standard firmata ethernet.

I've tried 4 different nanos, and a mega 2560 with 2 different bluetooth modules (HC-06), these never seem to connect and always result in the pin configuration not received error.

I've tested from my phone, tablet and laptop using both the remote arduino experience app and a custom remote blinky app.

I had noticed a few errors about the capabilities on the remote arduino experience, so I tried deploying the remote blinky app ensuring the bluetooth capability was selected.

All result in the same error.

I also tried this to no avail:

In file RemoteDevice.cpp
line 522:

                i += 4;
change to:

                i += 6;

And I tried this:

In RemoteDevice.cpp, around line 497, I changed Sleep( delay_ms ); to Sleep( 1000 );

This last one creates the error:

Connection attempt failed: BluetoothSerial::connectToDevice...
No more data is available.
)

I also saw a request time out error once.

Can anyone help?

Cheers,

Andy

mcinnes01, 2016-06-28:
Hi 

For anyone else experiencing this issue, I've increased the sampling interval and it seems to have resolved the problem with a nano using usb:

Line 67: unsigned int samplingInterval = 99;

Still getting the same error on bluetooth unfortunately. which I can't understand why?

I can see the arduino is doing stuff, but it never seems to manage to send the pin configuration through.

Andy

mcinnes01, 2016-06-29:
Hi,

Ok a little more investigating and I've got something working...

As I've said previously, I've got my cat feeder running on this with wifi, with all but the occasional disconnection that requires a manual reset. This is running an old version of standard firmata ethernet.

Using the arduino ide and its package manager I've removed and reinstalled the latest version of firmata: 2.5.2

With the and the standard or ethernet firmata example sketches, I get the problem described above.

However with the following sketch, it all runs fine. (This sketch is pretty standard but has a distance sensor sysex command added).

`/*
Firmata is a generic protocol for communicating with microcontrollers
from software on a host computer. It is intended to work with
any host computer software package.

To download a host software package, please clink on the following link
to open the download page in your default browser.

https://github.com/firmata/arduino#firmata-client-libraries

Copyright (C) 2006-2008 Hans-Christoph Steiner.  All rights reserved.
Copyright (C) 2010-2011 Paul Stoffregen.  All rights reserved.
Copyright (C) 2009 Shigeru Kobayashi.  All rights reserved.
Copyright (C) 2009-2015 Jeff Hoefs.  All rights reserved.

This library is free software; you can redistribute it and/or
modify it under the terms of the GNU Lesser General Public
License as published by the Free Software Foundation; either
version 2.1 of the License, or (at your option) any later version.

See file LICENSE.txt for further informations on licensing terms.

Last updated by Jeff Hoefs: April 11, 2015
*/

#include <Servo.h>
#include <Wire.h>
#include <Firmata.h>

#define ULTRASONIC_DISTANCE_QUERY	0x42
#define ULTRASONIC_DISTANCE_RESPONSE 0x43

#define I2C_WRITE                   B00000000
#define I2C_READ                    B00001000
#define I2C_READ_CONTINUOUSLY       B00010000
#define I2C_STOP_READING            B00011000
#define I2C_READ_WRITE_MODE_MASK    B00011000
#define I2C_10BIT_ADDRESS_MODE_MASK B00100000
#define MAX_QUERIES                 8
#define REGISTER_NOT_SPECIFIED      -1

// the minimum interval for sampling analog input
#define MINIMUM_SAMPLING_INTERVAL 10

/*==============================================================================
* GLOBAL VARIABLES
*============================================================================*/

/* analog inputs */
int analogInputsToReport = 0; // bitwise array to store pin reporting

							  /* digital input ports */
byte reportPINs[TOTAL_PORTS];       // 1 = report this port, 0 = silence
byte previousPINs[TOTAL_PORTS];     // previous 8 bits sent

									/* pins configuration */
byte pinConfig[TOTAL_PINS];         // configuration of every pin
byte portConfigInputs[TOTAL_PORTS]; // each bit: 1 = pin in INPUT, 0 = anything else
[... 688 lines cut ...]
	if (currentMillis - previousMillis > samplingInterval) {
		previousMillis += samplingInterval;
		/* ANALOGREAD - do all analogReads() at the configured sampling interval */
		for (pin = 0; pin < TOTAL_PINS; pin++) {
			if (IS_PIN_ANALOG(pin) && pinConfig[pin] == ANALOG) {
				analogPin = PIN_TO_ANALOG(pin);
				if (analogInputsToReport & (1 << analogPin)) {
					Firmata.sendAnalog(analogPin, analogRead(analogPin));
				}
			}
		}
		// report i2c data for all device with read continuous mode enabled
		if (queryIndex > -1) {
			for (byte i = 0; i < queryIndex + 1; i++) {
				readAndReportData(query[i].addr, query[i].reg, query[i].bytes);
			}
		}
	}
}`

The main thing I notice is the min sampling interval used to be 10, but this is now 1. Also the baud rate used to be 9600 and is now 57600.

Hope this helps someone else

Andy

codehArt, 2016-07-04:
hi,
i am trying to implement simple standard firmata wifi between arduino mkr 1000 and windews remote arduino, but when i am trying to connect from windows remote arduino experience it is giving me an error
#### connection attempt failed : adevice connection was established, but the device failed handshaking procedures.
#### verify that your device is configured with standard firmata.
#### message: catastrophic failure
#### pin configuration not received.

while simple standard firmata is working fine on usb port .
any one can help me regarding this issue.

carlosramajo, 2016-11-25:
Hi
I had the same problem and I tried everything during hours, but finally I solved my problem. My problem was the most stupid problem in world, I had connected the bluetooth TX-TX, RX-RX instead of TX-RX, RX-TX. :-( 
TIP: check the connections, please don't be like me

IoTGirl, 2017-03-20:
Closing based on solution to verify pin connections

closed 2017-03-20 by IoTGirl

Kulbhushan-Chand, 2019-01-31:
@IoTGirl This issue should not be closed. The main issue is the failed handshaking with Arduino Mega and it still persists.