ms-iot/remote-wiring #93: How to use I2C functionality

mcinnes01, 2016-03-09:
Hi,

I am trying to implement the BMP280 via i2c with an arduino, which I have connected up as follows and have had working with code native to the arduino and spark, but not via remote arduino:

https://learn.adafruit.com/adafruit-bmp280-barometric-pressure-plus-temperature-sensor-breakout/wiring-and-test

I have found an adafruit BMP280 example for raspberry pi, but haven't been able to get it to work with the remote arduino code.

https://github.com/ms-iot/adafruitsample/tree/master/Lesson_203/FullSolution

I've taken the blinky app which I have working over ethernet with an arduino, and have lifted the BMP280.cs from the adafruit example, and then tried reading the temperature.

I just wondered if you can give me any guidance as to how I can hook this up with the arduino to work via firmata?

Many thanks

Andy

jasonheg, 2016-05-09:
I too am having an issue working with I2C/Remote-Arduino. I'm using an HDC100x temp sensor, and can read the Device ID (0x1000) using the native Arduino Wire library commands.  However, when using the I2C event/callback functionality, I receive no response except for 0s.

To verify, I am having the callback function called, but no data is received using the DataReader variable.

I'm not sure if I'm doing it correctly, but here's the code I'm using:
`arduino.pinMode(20, PinMode.I2C);
arduino.pinMode(21, PinMode.I2C);
arduino.I2c.enable();
arduino.I2c.I2cReplyEvent += getData;

arduino.I2c.beginTransmission(0x40);
arduino.I2c.write(0xff);
arduino.I2c.endTransmission();
arduino.I2c.requestFrom(0x40, 2);

private void getData(byte myAdd, byte myReg, Windows.Storage.Streams.DataReader myRep)
        {

            var ignored = Dispatcher.RunAsync(Windows.UI.Core.CoreDispatcherPriority.Normal, () =>
            {
                textBox.Text = "Received: ";
                byte[] myVals = new byte[2];
                myRep.ReadBytes(myVals);
                foreach (byte i in myVals)
                {
                    textBox.Text += i;
                }
            });
        }`

I've tried putting the data gathering inside and out of async functions without any change.  I've also tried changing how the data is gotten from the myRep variable.  I'm guessing I might be missing something fundamental, but maybe someone can offer some additional guidance.

jasonheg, 2016-05-09:
OK, got it working with the UNO.  Perhaps I was having problems because I was trying to use the DUE and it's not quite fully supported?  Not sure, but I still receive zeros on the Due, and the Uno is still a little finicky, though I can receive the correct response.

I had to issue two requestFrom statements, but also needed to request more bytes than I actually needed to read out the register from my device.  For instance, when checking the ID, I only actually _needed_ to request 2 bytes to verify reading was correct.  Instead, I used a request for 18 bytes, twice.  This allowed me to click a button, and get the correct response from the register that had the device ID.  Without putting two identical requestFrom statements, I had to click the read button (That I programmed) multiple times before the correct response was obtained.

I went back and tried this with the Due, but it still only gave me zeros.  I also tried on the SDA/SCL(1) pins on the Due, with no difference in outcome.

So, perhaps a timing issue, or maybe something else, but there's at least one way to get it to work (On the Uno).  I think the Due Firmata might be problematic.  Other functionality does work on the Due, but not the I2C from what I can see.

Hope this makes sense.  Let me know if you still have problems.

zfields, 2016-05-18:
@jasonheg Does your other issue encapsulate this issue as well? @mcinnes01 does @jasonheg's solution remedy your issue?

zfields, 2016-05-18:
Ideally you would use remote-wiring to create a virtual instance of your Arduino and send the I2C transactions in the standard Arduino format.

jasonheg, 2016-05-18:
I wasn’t aware of an remote-wiring instance version of Wire.read.  I believe it used to show up, but no longer does.

For example, if I have an instance of remote-wiring called ‘arduino’, I can’t make a call to arduino.I2c.read() as I would make a call to wire.read() in Arduino.

Is there something similar?  Are you guys aware that the callback doesn’t specify the register upon reading it after a response?  Or am I just doing something wrong?

I even tried using a firmata call (Using .getRegister()), but still don’t see anything in the reg variable except for 0.

zfields, 2016-05-18:
@turkycat wrote the i2c functionality, so he is the best to answer your question. I'm on vacation, but if we don't hear from him before I get back, then I'll try and work out an answer for you.

mcinnes01, 2016-05-24:
Yes this is pretty much where I'm coming undone, the only details or examples I've seen reference methods that either no longer exist or are used in a different way. I did find some fairly helpful bit here: https://channel9.msdn.com/Events/Ignite/Australia-2015/MOB222 and here: https://github.com/mitchdenny/msftignite2015au-mob222.

IoTGirl, 2017-01-24:
Hi Jesse,
Could you please take a look at this issue?
Sincerely,
IoTGirl

lucora, 2017-08-26:
I've solved in this mode (I don't know how) : 
  
       private async void ButtonI2CReceive_OnClick(object sender, RoutedEventArgs e)
        {
            arduino.I2c.enable();
            arduino.I2c.requestFrom(9, 16);
            await Task.Delay(10);
            arduino.I2c.requestFrom(9, 16);
        }

I receive only one event on trigger it.

saraclay, 2018-08-21:
If this issue is still relevant, please file this at: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-08-21 by saraclay