ms-iot/remote-wiring #96: Guidance on disconnecting

jsiegmund, 2016-04-20:
Hi there, I'm going to crosspost this from https://social.msdn.microsoft.com/Forums/en-US/7728b0f0-2842-4da8-add5-950b11847417/best-practices-for-connecting-disconnecting-bluetooth-devices?forum=WindowsIoT, hoping to get some more info here.

I'm looking for some best practice guidance on connecting and in particular disconnecting bluetooth devices. The connecting part is all and well, working and everything. But I want to make my app (running on Raspberry Pi) resilient to connection failures too. When the connection to a bluetooth device is lost for whatever reason, I have some logic which tries to re-establish the connection using the same code that's used to set it up initially. Unfortunately that doesn't seem to work and I'm not sure how to improve it. 

I'm using classes of the following types: 

- DeviceInformation - which I get for my two paired blueooth devices. 
- BluetoothSerial - (from Microsoft.Maker.Serial), for serial communication
- UwpFirmata - (from Microsoft.Maker.Firmata), to implement the Firmata protocol
- RemoteDevice - for interacting with the actual device (Arduino btw) 

So what I'm looking for is: which classes do I dispose / make null / recreate, which do I maybe reuse? What methods should I use for reconnecting cause obviously my current attempts just recreating everything are not the way to go. I also tried disposing everything, which ended in internal exceptions being fired (sorry, don't have any stacktrace at the moment). I should note that that was a couple of builds ago, so I could retry it and see whether things have improved over time. 

I'm also seeing some weird behavior connecting and disconnecting devices. Sometimes my app will crash (mostly just bugs in my software, don't worry). If I now restart the app, in some occasions Windows has lost the DeviceInfo object for one of my paired bluetooth devices. I use the following code to query the device registry: 

`DeviceInformation.FindAllAsync(RfcommDeviceService.GetDeviceSelector(RfcommServiceId.SerialPort))`

Normally, this method returns two objects of type DeviceInfo. But when this behavior occurs, it will return only one. Now one might think the other device is in some sort of locked state which was not freed because the app crashed. I would agree, but even after restarting this issue still persists. Btw same happens when I exit the app from Visual Studio.

The only workaround I have at the moment it to unpair the device and pair again via the web interface. After re-pairing, above method returns two items again. So still on the look-out for some good advise on how to handle these connections! :)

zfields, 2016-05-18:
Thanks for reporting this!

saraclay, 2018-08-21:
If this issue is still relevant, please file this at: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-08-21 by saraclay