ms-iot/remote-wiring #97: I2C Reply / Register Reply

jasonheg, 2016-05-12:
When using I2C, aside from the issue of having to request several more bytes than needed, I can't seem to get a response as to what register is responding to the original I2C write command.

For instance, when using the below:
    arduino.i2c.beginTransmission(0x40);
    arduino.i2c.write(0x1f);
    arduino.i2c.endTransmission();
    
    arduino.i2c.requestFrom(0x40, 30);
    arduino.i2c.requestFrom(0x40, 30);

I can't seem to actually read what register is replying, though I do get the correct response.  The register address always reads as 0.  This makes it a bit difficult to process several different requests without creating several callbacks.

If I don't request 30 bytes twice (Even though I only need 2 bytes for the register address), I constantly have to click the button to initiate the request and then it eventually responds.  Perhaps it's because of this request for 30 bytes, but I don't know any other way to actually get the I2C to fire off the callback without having to hammer the button to keep asking.

Is there something I'm still fundamentally missing in regards to I2C?  I also tried using a firmata callback, but also receive 0 for the register that's replying.  In the above example, I would image 0x1f should be what is represented in the reg_ variable for the callback.

Please advise.  Thank you!

jasonheg, 2016-05-12:
Also, is there no longer a way to process I2C data manually rather than using callbacks?  It might be easier if it's how I recall it used to be (Where you can actually issue i2c.read commands), though it's been a little while since I last used this platform and my memory is a little foggy.

zfields, 2016-05-18:
@turkycat will you take a look at this?

mcinnes01, 2016-07-02:
Hi @turkycat 

Still struggling with I2C, are there going to be some clear instructions/examples of the recommended way of using it?

The only example I have seen is from an Australian MS Ignite video from 2015 and that used the callback event which is registered on page load.

The situation I'm trying to work with is to use button click events to read a sensor value, but I'm not sure the best way to approach this scenario or if there is a direct I2C call anymore as @jasonheg queried?

Any help, details, examples or documentation would be most welcomed :)

Many thanks

Andy

IoTGirl, 2017-01-24:
Hi Jesse,
Could you take a look at this issue?
Sincere thanks,
IoTGirl

saraclay, 2018-08-21:
If this issue is still relevant, please file this at: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-08-21 by saraclay