ms-iot/remote-wiring #105: Code not working in Visual Basic 2015

TonyScarpelli, 2016-07-02:
I am trying to get Windows-Remote-Arduino working in a Visual Basic 2015 Universal application.

I got it to work all right with this C# code:
  connection.ConnectionEstablished += OnConnectionEstablished;
  connection.begin(115200, SerialConfig.SERIAL_8N1);

In VB, when I type in  connection.  the options that appear are: available, begin, connectionReady, etc.

None of them seem to work.

I am trying to figure out the code to make this work.
What option is similar to ConnectionEstablished?

Thanks.

sadersa, 2017-01-07:
try
connection.DeviceConnectionFailed+= OnDeviceConnectionFailed;

IoTGirl, 2017-01-26:
No response - closing as stale.

closed 2017-01-26 by IoTGirl