ms-iot/remote-wiring #111: Reading from Arduino only after 99 bytes received

nambuco, 2017-01-22:
Hello All,

I am building an App using custom SysEx commands but before implementing the new SysEx I am testing using StandardFirmata just sending REPORT_FIRMWARE (0x79) SysEx and checking the answer. 

I have to send 2-3 time the SysEx to get the answer. After trying and investigating deeper, I saw that remote-wiring only read the input buffer when there is 99 bytes in the buffer. 

Is that the normal behavior? 

Is there a way to read byte from serial as they come?

I am using a Win10 and a Arduino UNO with StandardFirmata sketch.

Thanks in advance,

Paulo Henrique (nambuco)

nambuco, 2017-01-22:
After some teste I could solve the problems changing on USBSerial.h on line 135 from:
`	static const uint8_t MAX_READ_SIZE = 100;`
to
`	static const uint8_t MAX_READ_SIZE = 1;`

It seems the method LoadAsync(MAX_READ_SIZE) only resume execution after all MAX_READ_SIZE are read. Even with the InputStreamOptions as Partial (see line 461 from USBSerial.cpp). This happens in other codes I tested....

I am not sure if this solution will generate any side effect....

IoTGirl, 2017-01-27:
Hi Nambuco,
You have stated that you have addressed your problem so I am closing this issue.  Please reopen with more details if you disagree.
Sincerely,
IoTGirl

closed 2017-01-27 by IoTGirl