ms-iot/remote-wiring #112: "AnalogWrite" with incremental delay after using "sendSysex"

marcodiniz, 2017-02-07:
I'm getting a strange behavior.

I have this code running in a 50ms loop:
`                _firmata.sendSysex(0xAA, _buffer.AsBuffer());
                _firmata.sendSysex(0xAB, new byte[1].AsBuffer());`

Then o call bunches of 'analogWrite' to move a servo;
The first time the servo move instantly, after the secondo bunch of 'analogWrite' the commands start to delay, like it was buffered. The second command bunch delay approximately1 sec, the second 3 secs and go on until theu are taking minutes do got executed.

The breakpoint is hitting 'analogWrite' on the correct time.

Thanks!

IoTGirl, 2017-02-07:
Hi Marcodiniz,

I had members of the iot Team review this issue and you have not provided sufficient detail for them to understand how to see your issue.  Specifically:

- I can’t see anything obvious here, I took another look at the code to refresh my memory of the flushing patterns, everything looks good. The smoking gun here is that it “works from his pc and not his RPi”. He should be more specific what he means by this.

- I’d be curious how they are connecting it to the pi

Can you please provide far more detail as to your setup?  What hardware and software is involved (including versions and how they are connected)? How would someone at Microsoft see your issue? Please provide a full description to reproduce the issue.

Sincere thanks,
IoTGirl

marcodiniz, 2017-02-07:
Hi IoTGirl,

Of course.
I'm codding an UWP app for the raspberry, this app connect to an arduito using Firmata. I'm currently using UART0 to connect the Arduino and RPi3 at 57600 rate.
I'm also driving some NeoPixel leds, as they need a constant and fast frequency to work I had to create a custom command on StandarFirmata, this one:

`  switch (command) {
    case LEDSET:
    {
      byte led;
      byte r;
      byte g;
      byte b;
      for(byte i =0;  i< NUMLEDS;i++)
      {
        if(i*3>argc)
          return;
        r = argv[i*3];
        g = argv[i*3+1];
        b = argv[i*3+2];
        leds[i] = CRGB(r,b,g); 
      }
      FastLED.show();
      break;
    }`

This is working fine I guess, I also tried with other libraries for the same LEDs and the problem persists, I guess is something on UWP side.
On my code I have a timer to keep updating the LEDs, for fade effects and stuff, the timer "tickes" on every 50ms and update all my leds trought Sysex (12 leds x 3 bytes = 36bytes per call)
`        public async Task Flush()
        {
            byte r, g, b;
            r = g = b = 0;
            for (byte i = 0; i < All.Count; i++)
            {
                var led = All[i];
                if (led.Mode == 1)
                {
                    r = led.Color.RelativeR();
                    g = led.Color.RelativeG();
                    b = led.Color.RelativeB();
                }
                else if (led.Mode == 2)
                {
                    r = led.Color.RelativeR();
                    g = led.Color.RelativeB();
                    b = led.Color.RelativeG();
                }

                _buffer[i * 3] = (byte)(r / 2);
                _buffer[i * 3 + 1] = (byte)(g / 2);
                _buffer[i * 3 + 2] = (byte)(b / 2);
            }
            var iBuff = _buffer.AsBuffer();
            _firmata.sendSysex(0xAA, iBuff);
            _firmata.flush();
        }
`

Also, apart from the LED's animations, I'm driving some servos, I send a bunch of positions to get a nice accelleration. Example: Moving 180º will generate an array of positions and an array of delays for those positions. This is the code from my code to Firmata:

`
            //let's move
            for (int j = 0; j < totalSteps; j++)
            {
                await SendPosition(positions[j], delays[j]);
            }`
`        private async Task SendPosition(int pos, int delay = 10)
        {
            if (pos > MaxPos)
                pos = MaxPos;
            if (pos < MinPos)
                pos = MinPos;
            if (IsInverted)
                pos = (MaxPos - pos);

            await _pin.AnalogWrite((ushort)pos, delay);
        }
`
`        public async Task AnalogWrite(ushort value, int delay = CommandDelay)
        {
            Parent.Firmata.sendAnalog(Number.Value, value); <----THIS LINE CALLS THE FIRMATA!! Pin and Value
            AnalogValue = value;
            await Task.Delay(delay);
        }`

Supposing the LEDs is fading all the time, the problem is: When I manually (clicking a button) send a movement request the servo move instantly, after 2 or 3 requests the servo start delaying to move after the click. After 20 clicks it's getting a minute to move.

The strange thing is that if I put a breakpoint on the Firmata Call line, it get a bunch of hits (the positions array) but the servos don't move. If I wait some without clicking anything the servo will start to move.
But this happens only when the LED's animation is running, if I remove the Sysex call the servos start to respond immediatly

Here is my code:
[TinBotFirmata.txt](https://github.com/ms-iot/remote-wiring/files/759331/TinBotFirmata.txt)
[TinBot2EletronicValidation.zip](https://github.com/ms-iot/remote-wiring/files/759330/TinBot2EletronicValidation.zip)

saraclay, 2018-08-21:
If this issue is still relevant, please file this at: https://github.com/MicrosoftDocs/windows-iotcore-docs

closed 2018-08-21 by saraclay