ms-iot/remote-wiring #119: Multiple Arduinos (RemoteDevices)

dflintt, 2017-04-21:
Is it possible to have two arduinos hooked up a pi simultaneously? Using something like the following:

 connection1 = new UsbSerial("VID_2A03", "PID_0043");
 arduino1 = new RemoteDevice(connection1);

 connection2 = new UsbSerial("VID_2A03", "PID_0043");
 arduino2 = new RemoteDevice(connection2);

arduino1.DeviceReady += SetupAOne;
arduino2.DeviceReady += SetupATwo; 

connection1.begin(9600, SerialConfig.SERIAL_8N1);
connection2.begin(9600, SerialConfig.SERIAL_8N1);

Or is this functionality not possible? I've been trying the above without success, and I'm not sure if its because I'm doing something wrong or it just isnt possible with this library

IoTGirl, 2017-04-21:
Hi @dflintt,
This question was already asked on the forums at [https://social.msdn.microsoft.com/Forums/en-US/327c7893-e65b-4274-b5b9-4221328fc639/windows-remote-arduino-with-more-arduino-devices?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/327c7893-e65b-4274-b5b9-4221328fc639/windows-remote-arduino-with-more-arduino-devices?forum=WindowsIoT)
The code is only written to handle one Arduino but you can certainly attempt to extend it yourself.
Sincerely,
IoTGirl

IoTGirl, 2017-04-24:
Closing issue based on reference to answer given.

closed 2017-04-24 by IoTGirl