ms-iot/remote-wiring #120: Arduino Mega not working with WRAE

dflintt, 2017-04-21:
I've been trying to using an Arduino Mega (2560) with this board. I uploaded standard firmata, and tried using windows remote arduino experience to connect to the board, in order to test that firmata is working correctly. However, it tells me the connection has failed, and asks if I've uploaded standard firmata.

Is this a common issue? I figured that it should at least be working for windows remote arduino experience, if not when it is hooked up to the pi.

IoTGirl, 2017-04-21:
Hi @dflintt,
It seems editing the RemoteDevice.cpp is the best way to use Remote Arduino with the Mega.  This was discussed at [https://social.msdn.microsoft.com/Forums/en-US/33ee6f83-fab8-43c0-8d28-b85d8347634a/arduino-mega-and-firmata?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/33ee6f83-fab8-43c0-8d28-b85d8347634a/arduino-mega-and-firmata?forum=WindowsIoT)
Sincerely,
IoTGirl

dflintt, 2017-04-24:
Sorry, I don't think I was clear enough. I know that this has been discussed, but my problem is that I can't edit RemoteDevice. See picture below:

[image]

The installation guide says to install windows remote arduino using the nuget package manager command (link here: https://github.com/ms-iot/remote-wiring/blob/develop/installation.md ). This adds the reference in the project, but clicking on it pops up the error message above. (This project has already been built, so that isn't the issue).

The alternative installation instruction is to import the microsfot maker firmata and microsoft maker remotewiring packages (which contain remote device) into the project and them add them as a reference. But that doesn't appear to work, as it is giving me the following error:

[image]

So I suppose to rephrase my question, do you know why this isn't importing correctly anymore? I followed the manual install instructions on the link above.

dflintt, 2017-04-25:
@IoTGirl I don't mean to be rude, but this is fairly urgent, and a response would really be appreciated.

DeveloperMK, 2017-05-14:
Sleep ( 1000 ); does not work for me but Sleep( 200 ); works.

IoTGirl, 2017-05-16:
Closing based on answer given by @DeveloperMK

closed 2017-05-16 by IoTGirl