ms-iot/remote-wiring #122: remote-wiring in not compatible with Arduino Mega

Kenuat, 2017-05-14:
I'm having the same issue, as described in https://github.com/ms-iot/remote-wiring/issues/120 and https://github.com/ms-iot/remote-wiring/issues/80, my Mega fails to initialise. I'm using USB connection with 57600 baud rate, same happens when runing my code or MS app from Store.

And the most confusing thing here is that when remote-wiring starts its discovery, arduino status led becomes active for around 17 seconds. At around 10th second app fails with the "pin configuration not recieved" error.
But sometimes (very rarely though) it takes around only one second to initialise! Without any code change, just a simple rerun. And if that happens, then my code works well in that case. I couldn't find any dependencies in this behaviour.

The full error message from failed runs debug output:

> Exception thrown at 0x76E1A9F2 (KernelBase.dll) in robot_sharp.exe: 0x40080201: WinRT originate error (parameters: 0x8000FFFF, 0x0000001F, 0x016FF608).
> Exception thrown at 0x76E1A9F2 in robot_sharp.exe: Microsoft C++ exception: Platform::Exception ^ at memory location 0x016FFB04. HRESULT:0x8000FFFF
> WinRT information: Pin configuration not received.

Successful run messages (while app is still running):

> The thread 0x4084 has exited with code 0 (0x0).
> The thread 0x2d48 has exited with code 0 (0x0).
> The thread 0x50ac has exited with code 0 (0x0).

But there is more: I can run any Firmata code using libraries written in other languages: firmata_test.exe application from firmata and very trivial python code using pymata module:
```import asyncio
from PymataCore import PymataCore

async def main():
    while True:
        await board.digital_write(13, 0)
        await asyncio.sleep(1)
        await board.digital_write(13, 1)
        await asyncio.sleep(1)

board = PymataCoreFixed(com_port="COM3")
loop = asyncio.get_event_loop()

loop.run_until_complete(board.start_aio())
asyncio.ensure_future(main())
loop.run_forever()```

This code works with the same board and it takes around one second for pymata library to make board discovery and print this string "Auto-discovery complete. Found 70 Digital Pins and 16 Analog Pins"

Taking in account that board works correct using other firmata libraries and event works with remote-wiring sometimes (very rarely), I'm totally sure that there is a bug in remote-wiring somewhere which makes it incampatible with Mega.

Thanks.

DeveloperMK, 2017-05-14:
im using mega with bluetooth module HC-05 there is no problem for now. i just changed delay in RemoteDevice.cpp

IoTGirl, 2017-05-16:
Hi @Kenuat,

This sample was created with an UNO as pictured.  Other devices will likely need tweaking for timing and hardware differences. The repository is available for you as suggested by @DeveloperMK and many other forums have the same suggestion to modify RemoteDevice.cpp to meet your need.

I am closing this issue based on @DeveloperMK correct response.

Sincerely,
IoTGirl

closed 2017-05-16 by IoTGirl