ms-iot/remote-wiring #124: Could not install package 'Windows-Remote-Arduino 1.4.0

kyogre94, 2017-05-23:
Could not install package 'Windows-Remote-Arduino 1.4.0'. You are trying to install this package into a project that targets '.NETFramework,Version=v4.6', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.

Any idea how to fix this?

bfjelds, 2017-05-24:
Using VS2017, I was able to load Microsoft.Maker.win10\Microsoft.Maker.sln.  

I see a couple problems with project references to Microsoft.Maker.Serial in:

    Microsoft.Maker.win10/Microsoft.Maker.Firmata/Microsoft.Maker.Firmata.vcxproj
    Microsoft.Maker.win10/Microsoft.Maker.RemoteWiring/Microsoft.Maker.RemoteWiring.vcxproj

I updated both projects to reference:

    ..\..\source\Serial\Microsoft.Maker.Serial.win10\Microsoft.Maker.Serial.vcxproj

I also retargeted the projects to ensure that they were targeting what I had installed on my machine.

After that, I had no problem building.  

I could also use Manage Nuget Packages to download Windows-Remote0Arduino 1.4.0.

Can you provide more details about how you are seeing this issue?

kyogre94, 2017-05-25:
I solved the issue by installing the package for a universal windows app. It doesn't seem to install for a windows form application project.

Yes I was using the 'manage nuget packages' option to download Windows-RemoteArduino 1.4.0

bfjelds, 2017-05-25:
Glad you got it working!

closed 2017-05-25 by bfjelds