ms-iot/remote-wiring #126: Analog Write Doesn't work in v1.4

ankur198, 2017-06-04:
I don't understand the working but I tried in my project that when I used remote-arduino v1.4 from nuGet I wasn't able to perform analogWrite operations, but after downgrading to v1.2 (without any other code chamges), everything worked perfectly.

IoTGirl, 2017-06-06:
Hi @ankur198,
Can you tell us more about your setup? If IU were to try and see your issue what devices, software and hardware would I need and what steps would I have to take to see this issue. 
NOTE: Saying to use the steps in GITHUB is not enough. Please clarify which step and how the failure appears.
Sincerely,
IoTGirl

ankur198, 2017-06-09:
Actually that was my fault, I tried to analog write on a pin where it was not possible. Version 1.4 didn't sent a message to arduino at all, however version 1.2 did sent it.

closed 2017-06-09 by ankur198