ms-iot/remote-wiring #136: StringMessageReceived event not always firing

pireef, 2018-09-05:
Originally I posted this on [stackoverflow](https://stackoverflow.com/questions/52157002/windows-remote-arduino-stringmessagereceived-event?noredirect=1#comment91303943_52157002) .  After my issue has been reproduced it was suggested I open an issue here.  

So I am working on a project which utilizes Windows Remote Arduino. I'm just starting so I am in the "just hack it together stage." The arduino (Nano) is running standard firmata and I'm calling a sysex function which returns some data from some sensors.

I've subscribed to the StringMessageReceived Event, but what I'm finding is that event is not always firing. I 'should' be receiving data each time I send the sysex command, but I'm not getting a 1 for 1 response.

I and others have tested with a simple string message from the arduino with similar results.  

Here's a sample output I'm getting.

Device Ready Event Fired

Timer Tick!

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

Timer Tick!

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

Timer Tick!

Timer Tick!

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

Timer Tick!

Timer Tick!

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

Timer Tick!

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}

Timer Tick!

Timer Tick!

Timer Tick!

{\"temp1\":21.50,\"pH\":0.000}"

I know the arduino is getting my commands, because I can see the LED's on one of my sensors processing the command.

Here's my UWP code.  

public sealed partial class MainPage : Page
{
    IStream connection;
    RemoteDevice arduino;
    UwpFirmata firmata;
    DispatcherTimer timer;

    public MainPage()
    {
        this.InitializeComponent();
    }

    private void btnConnect_Click(object sender, RoutedEventArgs e)
    {
        connection = new UsbSerial("VID_0403", "PID_6001");
        firmata = new UwpFirmata();
        arduino = new RemoteDevice(firmata);
        timer = new DispatcherTimer();
        firmata.begin(connection);

        connection.begin(57600, SerialConfig.SERIAL_8N1);
        arduino.DeviceReady += Setup;
        arduino.DeviceConnectionFailed += ConnectFailed;
        arduino.StringMessageReceived += StringRecieved;

        timer.Interval = new TimeSpan(0, 0, 5);
        timer.Tick += timerTick;
        timer.Start();
    }

    private void timerTick(object sender, object e)
    {
        Windows.ApplicationModel.Core.CoreApplication.MainView.CoreWindow.Dispatcher.RunAsync(Windows.UI.Core.CoreDispatcherPriority.Normal, ()
        =>
        {
            lblPhData.Text += "Timer Tick!\n";
        }
        );
        byte PH_QUERY = 0x44;
        firmata.sendSysex(PH_QUERY, new byte[] { }.AsBuffer());
        firmata.flush();
    }

    private void StringRecieved(string message)
    {
        Windows.ApplicationModel.Core.CoreApplication.MainView.CoreWindow.Dispatcher.RunAsync(Windows.UI.Core.CoreDispatcherPriority.Normal, ()
        =>
        {
            lblPhData.Text += message + "\n";
        }
        );
    }

    public void ConnectFailed(String message)
    {
        Windows.ApplicationModel.Core.CoreApplication.MainView.CoreWindow.Dispatcher.RunAsync(Windows.UI.Core.CoreDispatcherPriority.Normal, ()
             =>
         {
             lblPhData.Text += message;
         }
        );
    }
    public void Setup()
    {
        Windows.ApplicationModel.Core.CoreApplication.MainView.CoreWindow.Dispatcher.RunAsync(Windows.UI.Core.CoreDispatcherPriority.Normal, ()
             =>
         {
             lblPhData.Text += "Device Ready Event Fired\n";
         }
        );            
    }

    private void btnGetData_Click(object sender, RoutedEventArgs e)
    {
        byte PH_QUERY = 0x44;
        firmata.sendSysex(PH_QUERY, new byte[] { }.AsBuffer());
        firmata.flush();
    }
}

luyaok, 2018-09-06:
I can reproduce this issue on both Windows IoT Core and Desktop.

saraclay, 2018-11-09:
Hello,

If this is still an issue for you, could you please repost this in our MSDN forums? We have staff there that are monitoring the forums and will be able to help you.

MSDN forums: https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

closed 2018-11-09 by saraclay